esa-tu-darmstadt/tapasco #294: Probably bug in ip.tcl

forflo, 2021-07-15:
https://github.com/esa-tu-darmstadt/tapasco/blob/8b21b7fa7d0c4142402554425b5b9b337d45d20f/toolflow/vivado/common/ip.tcl#L539

This line blows up when I enter a composition with more than eight different PEs. (For example `[ pe0 x 1, pe1 x 1, ..., pe8 x 1 ]`.

For `target_ip_08` the `regsub` command will expand to "08" wich will then be fed into `format`, which in turn will error out with `expected integer but got "08"`.

forflo, 2021-07-15:
In `develop` the same snippet is here: https://github.com/esa-tu-darmstadt/tapasco/blob/466467a2f4da08f9ff4535e6103942066b9f6ecf/toolflow/vivado/common/ip.tcl#L539

cahz, 2021-07-19:
Thanks for pointing this out. Leading zeros are treated as octal numbers in tcl...

Will push a fix in a few minutes.

PR #295 (2021-07-20): Correctly convert kind count to decimal number
Closes #294
closed 2021-07-20 by wirthjohannes