esa-tu-darmstadt/tapasco #308: Enable HBM with the --features option

jeudine, 2021-11-18:
Hi,
Could you please provide an example on how to enable HBM with the `--features` option during the compose job.
Since the `--features` option only accept the following syntax: `<NAME> <BEGIN> [<KEYVALUE> [, <KEYVALUE>]*] <END>`, how would it be possible to use it with the HBM example provided in  `tapasco/misc/documentation/tapasco-features.md `

StaCu, 2021-11-18:
Hi,
enabling HBM directly from the CLI is not possible.
Instead, you have to create/edit a config file manually and feed it to tapasco using
`tapasco --configFile config.json`

To create a config file, call
`tapasco compose [arrayinit x 2, arrayupdate x 2, arraysum x 2]  @ 50MHz -p AU280 --skipSynthesis --deleteProjects false`
a config.json will be created nested in the <tapasco_workdir>/compose/.. directory.
Edit the config.json by inserting the HBM feature manually as shown below.

{
   ...
    "Platforms" : [ "AU280" ],
    "Features" : [ {
      "Feature" : "HBM",
      "Properties" : {
        "hbm0" : {
          "ID" : "arrayinit",
          "Count" : "2",
          "Interfaces" : "m_axi_maxi_arr"
        },
	"hbm1" : {
          "ID" : "arrayupdate",
          "Count" : "2",
          "Interfaces" : "m_axi_maxi_arr"
        },
	"hbm2" : {
          "ID" : "arraysum",
          "Count" : "2",
          "Interfaces" : "m_axi_maxi_arr"
        },
	"enabled" : "true"
      }
    } ],
    "SkipSynthesis" : true,
    "DeleteProjects" : false
  } ]
}

Interfaces is a whitespace seperated list of the interfacenames as defined by the IPXACT-Core (in case the core has more than one memory interface)

wirthjohannes, 2021-11-18:
Just a quick addition to StaCu's answer as it has lead to confusion in the past:
The names "hbm0", "hbm1", etc. can be chosen arbitrarily. They do **not** influence the actual result. The example given would lead to the following results:
HBM <-> arrayinit
HBM <-> arrayinit
HBM <-> arrayupdate
HBM <-> arrayupdate
HBM <-> arraysum
HBM <-> arrraysum
(each "HBM" is a separate memory with a capacity of 256MB)

jeudine, 2021-11-18:
Thanks a lot

closed 2021-11-18 by jeudine