esa-tu-darmstadt/tapasco #329: Error during DMA operation with Fork systemcall in pcie-based FPGA

sajjadtamimi, 2022-06-28:
On the API-side, when I call tapasco.copy_to() with 64KB writes in a loop and fork and run it concurrently, I get the following error:

ERROR tapasco::ffi] Setting LAST_ERROR: Error during DMA operation: Got interrupt but outstanding buffers are empty. This should never happen.
Error during DMA operation: Got interrupt but outstanding buffers are empty. This should never happen.
terminate called after throwing an instance of 'tapasco::tapasco_error'
  what():  Error during DMA operation: Got interrupt but outstanding buffers are empty. This should never happen.

To have a better understanding, I made the 64KB write in a loop, till I get a similar error. Please find the RUST_LOG in here: 

[2022-06-27T21:31:32Z TRACE tapasco::dma_user_space] Copy Host(0xaaaab91e9cc0) -> Device(0x5ac30000) (65536 Bytes)
[2022-06-27T21:31:32Z TRACE tapasco::dma_user_space] Copy Host(0xaaaab91e9cc0) -> Device(0x5ac40000) (65536 Bytes)
[2022-06-27T21:31:32Z ERROR tapasco::ffi] Setting LAST_ERROR: Error during DMA operation: Got interrupt but outstanding buffers are empty. This should never happen.
Error during DMA operation: Got interrupt but outstanding buffers are empty. This should never happen.
terminate called after throwing an instance of 'tapasco::tapasco_error'
  what():  Error during DMA operation: Got interrupt but outstanding buffers are empty. This should never happen.

I also try to create a new process with thread() call, and there was no issue

sajjadtamimi, 2022-06-28:
Tapasco device handle is defined as static in a .cpp file and accessed through some C wrapper functions defined in the corresponding header file. This is then statically linked and used to access the tapasco copy_to functionality.

cahz, 2022-07-06:
Using shared memory should work fine with the TaPaSCo runtime.

closed 2022-07-08 by cahz