esa-tu-darmstadt/tapasco #332: No compatiblilty with JDK 14 and newer

tsmk94, 2022-08-18:
On Fedora 36 (openjdk 17.0.3) I got the following error message when trying to execute any tapasco command (even ```tapasco --help```):
```
Unrecognized VM option 'CMSClassUnloadingEnabled'
Error: Could not create the Java Virtual Machine.
Error: A fatal exception has occurred. Program will exit.
```
According to [here](https://stackoverflow.com/questions/70214788/unrecognized-vm-option-cmsclassunloadingenabled-after-switching-to-java-17) this is caused by the ```-XX:+CMSClassUnloadingEnabled``` option which is no longer supported since the CMS Garbage Collector has been removed in JDK version 14. However, just removing this option from our ```build.gradle``` does not solve the issue for me as now the build already fails with ```java.lang.IllegalArgumentException: Unsupported class file major version 61``` (-> incompatiblity of used gradle version with newer JDK?).

c-93, 2023-02-08:
To add some context:

[Stackoverflow](https://stackoverflow.com/questions/3334911/what-does-jvm-flag-cmsclassunloadingenabled-actually-do)

Some [oracle java documentation](https://docs.oracle.com/en/java/javase/11/tools/java.html#GUID-3B1CE181-CD30-4178-9602-230B800D4FAE):

> -XX:+CMSClassUnloadingEnabled
> 
>     Enables class unloading when using the concurrent mark-sweep (CMS) garbage collector. This option is enabled by default. To disable class unloading for the CMS garbage collector, specify -XX:-CMSClassUnloadingEnabled.

I guess, we can remove this flag. For me, this fixes the bug.
@tsmk94 : Could you review the MR?

c-93, 2023-02-08:
I also faced the error `java.lang.IllegalArgumentException: Unsupported class file major version 61`
However, clearing all build artifacts (cloning and building tapasco into a new folder), the error was gone.

cahz, 2023-02-08:
The issue is in the gradle-wrappe. jar provided in our repo. Upgrading this to 7.6 makes it compatible to newer java version.

PR #346 (2023-02-08): Java: Fix compatibility with JDK 14 and newer
Fix #332
This option is no longer needed for garbage collection: "-XX:+CMSClassUnloadingEnabled"
commit 907d2128 (2023-02-08): Java: Fix compatibility with JDK 14 and newer

Fix #332
This option is no longer needed for garbage collection: "-XX:+CMSClassUnloadingEnabled"
closed 2023-03-22 by cahz