esa-tu-darmstadt/tapasco #344: Multi-PE compositions on Ultra96 do not work

tsmk94, 2023-01-11:
Description
-------
In compositions with at least two PEs reads and writes from/to DDR fail with ```SLVERR``` error response. Additionally error messages such as the following are prompted in ```dmesg```:
```
[ 1979.500921] arm-smmu fd800000.iommu: Blocked unknown Stream ID 0x201; boot with "arm-smmu.disable_bypass=0" to allow, but this may have security implications
[ 1979.520654] arm-smmu fd800000.iommu:         GFSR 0x80000002, GFSYNR0 0x00000000, GFSYNR1 0x00000201, GFSYNR2 0x00000000
```

Possible Source
-----------
The interconnect tree used for multi-PE compositions adds IDs other than 0 to the read/write requests. These seem to trigger the SMMU faults (probably this has some other meaning for the SMMU).

Possible Solution
------------
Force request IDs always to zero, e.g. in the HPC-Offset core.

cahz, 2023-01-13:
Setting request IDs to 0 basically disables support for outstanding transactions. This will have a performance impact, so I would prefer a solution which configures the smmu properly.

tsmk94, 2023-01-13:
I don't see how it disables support for multiple outstanding transactions. You can issue transactions with the same ID without waiting for the preceeding request to be completed. However, it indeed disables reordering of read transactions.

I agree it might be useful to investigate how the SMMU could be configured properly.

tsmk94, 2023-02-08:
Fixed in ```develop``` branch

closed 2023-02-08 by tsmk94