esa-tu-darmstadt/tapasco #353: Compose broken for VC709 on develop

yannickl96, 2023-05-05:
During a compose job on VC709 I encountered the following critical warning:
```
CRITICAL WARNING: [Netlist 29-180] Cell 'MMCME3_ADV' is not a supported primitive for virtex7 part: xc7vx690tffg1761-2.  Instance 'system_i/memory/design_clk_wiz/inst/mmcme3_adv_inst' will be treated as a black box, not an architecture primitive [/scratch/yl/tapasco-upstream-workspace/compose/axi4mm/vc709/piccolo32_pe/001/050.0/microarch/microarch.gen/sources_1/bd/system/ip/system_design_clk_wiz_0/system_design_clk_wiz_0_clk_wiz.v:145]
Resolution: Modify instantiation of primitive to match Xilinx primitive. Please review the guidelines in the Design Methodology Guide UG949 to replace this primitive with a supported primitive.
```
which resulted in the following error later on:
```
CRITICAL WARNING: [Project 1-486] Could not resolve non-primitive black box cell 'MMCME3_ADV' instantiated as 'system_i/memory/design_clk_wiz/inst/mmcme3_adv_inst' [/scratch/yl/tapasco-upstream-workspace/compose/axi4mm/vc709/piccolo32_pe/001/050.0/microarch/microarch.gen/sources_1/bd/system/ip/system_design_clk_wiz_0/system_design_clk_wiz_0_clk_wiz.v:145]
Netlist sorting complete. Time (s): cpu = 00:00:00.02 ; elapsed = 00:00:00.02 . Memory (MB): peak = 4299.176 ; gain = 0.000 ; free physical = 11000 ; free virtual = 38558
open_checkpoint: Time (s): cpu = 00:00:53 ; elapsed = 00:00:57 . Memory (MB): peak = 4299.176 ; gain = 963.500 ; free physical = 10943 ; free virtual = 38502
Command: opt_design
Attempting to get a license for feature 'Implementation' and/or device 'xc7vx690t'
Running DRC as a precondition to command opt_design

Starting DRC Task
ERROR: [DRC INBB-3] Black Box Instances: Cell 'system_i/memory/design_clk_wiz/inst/mmcme3_adv_inst' of type 'MMCME3_ADV' has undefined contents and is considered a black box.  The contents of this cell must be defined for opt_design to complete successfully.
ERROR: [Vivado_Tcl 4-78] Error(s) found during DRC. Opt_design not run.

Time (s): cpu = 00:00:04 ; elapsed = 00:00:02 . Memory (MB): peak = 4371.211 ; gain = 64.031 ; free physical = 10953 ; free virtual = 38576
0 Infos, 1 Warnings, 2 Critical Warnings and 2 Errors encountered.
opt_design failed
ERROR: [Common 17-39] 'opt_design' failed due to earlier errors.
```

I am running TaPaSCo from the develop branch with Vivado 2022.1. The issue is introduced at some point between 228af47dc4f4803e7997b8ded7e740bde362359e and the latest commit.

yannickl96, 2023-05-05:
Running right now, I will keep you updated.

cahz, 2023-05-05:
It seems that 8f56b078a97a6320c478a97ba7f1c2d42f55374c causes Vivado to generate a wrong MMCM primitive for Virtex-7

closed 2023-05-05 by cahz