esa-tu-darmstadt/tapasco #372: Smaller bitwidth types trigger runtime error

yannickl96, 2023-08-18:
Using an ``int16_t`` value as argument to a job launch triggers the
```
terminate called after throwing an instance of 'tapasco::tapasco_error'
  what():  Please supply large arguments as wrapped pointers.
```
runtime error. I would suggest to either support smaller length types, i.e., ``(u)int{8,16}_t`` or provide a clearer error message, that for _normal_ arguments, only 32 and 64 bit are accepted and anything larger should be passed as a wrapped pointer.

tsmk94, 2024-02-13:
We only support 32 bit and 64 bit single arguments in the Rust runtime. So we could either add smaller types in the Rust runtime, or cast everything smaller than 32 bit to 32 bit values in the C++ header (which could lead to undesired behaviour either during cast or because strobes are not set as expected).

yannickl96, 2024-02-13:
I am actually quite agnostic if we have smaller types (16- or 8-bit). I just found the error message quite weird to debug and originally thought something was wrong with my buffer-based arguments. Since I am using the C++ API, the message probably originates from https://github.com/esa-tu-darmstadt/tapasco/blob/develop/runtime/libtapasco/src/tapasco.hpp#L291, so just changing the if-Statement should suffice to at least provide a clearer error message. If you are fine with that I can push that.

yannickl96, 2024-02-13:
Ok we will stick with a better error message.

PR #380 (2024-02-14): Provide better error message for small bitwidth arguments
This PR fixes #372. We now check in the C++ API if the provided argument type is smaller than 32 bit and tell the user quite clearly what is wrong.
closed 2024-02-14 by yannickl96